amund7/ScanMyTesla #4: Split UI between trips and real-time data

amund7, 2017-09-11:


amund7, 2017-09-18:
Perhaps the different trips could be a pulldown instead?

How about three tabs:
Logging - DIsplay - Trips
and then each one has it's own set of tabs, thru pulldown menu
Logging just means a seperate set of items, focused on making interesting logs, while Display will be what looks cool live

amund7, 2017-10-30:
I need help thinking this through/designing the UI. How should it look?

kdjcollins, 2017-10-31:
Amund, I have strictly been using real time data (20 parameters) and I haven’t used trips. I like logging and seeing 18 of the 20 variables on full screen.

amund7, 2018-09-18:
Had a new idea on this. Trips could be 1 tab, listing all trips as text, live updated with distance, maybe a start time etc. Then you could quickly browse or delete all trips, click one and you get the full page (with charge info etc).