amund7/ScanMyTesla #12: App lockup- communication loss maybe

kdjcollins, 2017-09-18:
Amund - occasionally (every 20-60 minutes of continuous use) app freezes (display unchanging) while logging. Post-run analysis also shows the logging to csv stopped. I use the recommended odb2 mx device on a Samsung Galaxy Note 3 phone. The status lights on the obd2 device seem to indicate comm loss, but phone shows Bluetooth connection to device. Resolution is unplug/plug-in obd2 device AND close/restart app. Is it possible this is due to logging too many signals? Or unrelated comm loss?

amund7, 2017-09-18:
I've seen that sometimes with my OBDLink MX as well - I thought it was too old :) Or just overwhelmed by the Tesla canbus traffic. To test if it's the adapter that locks up/crashes, try restarting the app, without re-plugging the adapter. If it gives errors/timeouts then it's the adapter. If it works again, there is a bug in my app, or I need better code to handle dropouts.

I managed to update the firmware in my OBDLink MX just the other day, I am hoping that will fix this hangup.

kdjcollins, 2017-09-18:
I have tried just killing then restarting the app each time the freeze up occurs, to no avail. Then I have unplugged / replugged the adapter and it usually restarts without errors. A couple of times I have had to restart the app AFTER unplugging / replugging the adapter. The adapter is brand new several weeks ago. I was displaying the "Perf" screen each time the lockup occurred (and was logging to csv file).

amund7, 2017-09-18:
I see. They have their own app called just 'OBDLink' - it will update to the latest firmware for you, but only if you connect the adapter to a gas or diesel car... I haven't had any hangups since (but also, I've seen very few of them, can't guarantee they've stopped.)

kdjcollins, 2017-09-18:
I have that OBDLINK app on my phone but haven't used it. I can see if the device needs an update.

kdjcollins, 2017-09-19:
[image]. I was on 4.3.0 and updated to latest 4.3.2.  We'll see if it helps.

kdjcollins, 2017-09-19:
I updated MX from 4.3.0 to 4.3.2. Will test over the next few days and see if it helps to eliminate freezing app.

kdjcollins, 2017-09-19:
App still freezes while logging to csv file, even after updating the OBDLINK MX firmware. Log attached as xlsx file as can't attach csv. screenshot of display also attached.
[Perf 2017-09-19 05-30-55.xlsx](https://github.com/amund7/ScanMyTesla/files/1314814/Perf.2017-09-19.05-30-55.xlsx)

[image]

amund7, 2017-09-19:
Can you still scroll, click, and stop the logging, or is the app completely frozen?

kdjcollins, 2017-09-19:
The app seems completely frozen, no response to touchscreen input. Have to press/hold home button, then flick app closed. Also have to unplug/replug OBDLINK MX device.

kdjcollins, 2017-09-20:
I checked that the phone (Galaxy Note 3) has the latest operating system software (AT&T) and I deleted suspected antivirus software app and battery power saver app, rebooted phone and retried data logging. App froze again after 15 minutes. Closing app and reopening app does not resolve - app can't start back up. I observed that the outer 2 LED's on the OBDLINK MX device were illuminated Power (green) and "BT" Bluetooth )blue), however the inner 2 LED's were NOT lit: "OBD" and "HOST". Only after unplugging then replugging OBDLINK device 2x was the app able to start up and begin logging again. All LED's lit. This time logging worked through test completion for about 50 minutes - no lock up. I am curious if issue has to do with peak detecting (min/max) of most of the "PERF" parameters being selected. Back before I learned how to invoke peak detection & display I don't recall getting this freezing up of the app. But maybe I don't precisely recall when I started that.

kdjcollins, 2017-09-20:
Tested this time without selecting peak detecting min/max values for any parameter on "PERF" display, while logging to csv file. Freeze up of My Tesla Scan app occurred after about 25 minutes of logging. After the freeze occurred, the OBDLINK MX adapter again had the outer 2 LED's on the OBDLINK MX device illuminated Power (green) and "BT" Bluetooth (blue), however the inner 2 LED's were NOT lit: "OBD" and "HOST".  This seems to indicate that the Bluetooth connection between OBD adapter and phone is still present but the app is frozen. This time instead of trying to resolve by killing then restarting app, which has been shown not to resolve the issue, I instead unplugged and replugged the OBDLINK MX adapter. The app remained frozen, until such time as I killed then restarted the app.

amund7, 2017-09-20:
You are overthinking it. The min/max value is constantly recorded any time any reading comes in, when you select something it prints on the screen, but it shouldn't make the app behave any differently.

I was thinking the cpu or disk of your phone could be overwhelmed/bogged down, but as you have such a new and powerful device that is very unlikely. I've also recorded lots of trips, and this problem has occured very rarely.

I'm not sure how, but if you could find a specific way to make the app hang like this then I could reproduce it. If you could also analyze the end of the logs right before the crash, maybe we can get some hints? Please send me the end of those logs, both CSV and RAW if you can.

Keep making it crash and see if you can find any common denominators. Try a different phone as well maybe? Could the disk be full, I don't think I ever tested that scenario.

kdjcollins, 2017-09-20:
The phone is my older backup Samsung - only Android - as I have iPhones. 2013 build. But checking specs and available memory looks OK (attached). It's still an operational (ATT) phone w sim card and number used just for car. I wondered if OBDLINK MX settings got changed when I tested it running "Torque" app on my Porsche. But the first Freeze fault began occurring on 9-12-2017 before I plugged device into any of the ICE cars. I will log both csv and raw files tonight on drive home- I am sure that it will freeze in that 1 hour drive. (I was thinking of doing a test sometime just displaying data without logging, when I have never done, just to see if it's related to saving data.)
[image]
[image]
[image]

kdjcollins, 2017-09-21:
I logged the long evening commute with no freeze up. A "watched pot  doesn't boil." Prior to the test, I did delete some bloatware and other unneeded apps from my phone, but flash memory wasn't full anyway, so this "success" may be coincidental and not statistically valid. I logged csv and raw files. CSV file is too big to open in excel, and I believe this site won't allow me to attached as csv, only excel. I will attach the raw file. So much for that, it wont allow files over 10MB.

amund7, 2017-09-21:
My mistake, I thought the note 3 was the brand new model, but that's called note 8 maybe?

kdjcollins, 2017-09-21:
Yes the latest is the Note 8, I believe. I logged another 30 minute run in this morning (logging both raw and csv) on "PERF" page with no app freezing issue. Just big file that excel can only display just over the first 1M rows-  excel limitation. Log looks good.

amund7, 2017-09-21:
Have you tried https://github.com/amund7/UDPLogger/releases/latest to graph CSV's? Works on pretty big ones as well, I've tried 50 mb ones.

kdjcollins, 2017-09-21:
I downloaded it earlier today and wasn't sure how to start, I found the setup.exe file thinking that would do the trick and it must not be the right way to start.

kdjcollins, 2017-09-22:
After several successful logging sessions of varying length (up to one hour each), this morning's logging session again caused what I have been calling "app freezing". Further investigation showed the following: (1) app in NOT frozen, but displayed parameter value as frozen at last read values. The screen DOES respond to touch input, it's just that the displayed numbers don't change, (2) Bluetooth communication LED on OBDLINK MX is still "ON" and the phone also shows Bluetooth is still "ON" with MX paired, (3) The MX's "OBD" and "HOST" LED's are "OFF", indicating no communication. This occurred just 20 minutes or so into the drive, so file size is much smaller than previous successful logging sessions, so probably no issue with filling up a buffer (certainly the phone has plenty of available memory - I cleaned out old logs yesterday). The MX device is placed less than 50 cm from the phone, with no metal obstructions, so unlikely a weak signal. Question what to try next (a) different (newer) phone, (b) different OBD device settings - revert to default - in case settings got changed by plugging in to ICE car OBD2 port, (c) test old OBD2/Bluetooth device (PLX).
[Perf 2017-09-22 05-32-19.xlsx](https://github.com/amund7/ScanMyTesla/files/1325236/Perf.2017-09-22.05-32-19.xlsx)

[RawLog 2017-09-22 05-32-20.txt](https://github.com/amund7/ScanMyTesla/files/1325231/RawLog.2017-09-22.05-32-20.txt)

amund7, 2017-09-22:
I think that closes the issue, this is the adapter hanging, and not the app.

It fits with my experiences, adapter hangs (only the OBDLink MX, not the other adapters I have), and there is no way to get back to working condition without unplugging the adapter. Just restarting the app will not help, so I don't think there's much I can do in the app as far as error detection/restarting bluetooth etc.

Closed.