amund7/ScanMyTesla #9: No front rpm reported

kdjcollins, 2017-09-18:
Amund, No front rpm is reported in the csv file, yet the column with front rpm heading is present. I realized that the signal is not on the "Perf" display, so maybe it is just not configured. Note that front torque and front hp are showing properly on both csv file and on display. (I am "Kenneth Collins" on facebook (Tesla Worldwide group), kdjcollins on google). Thanks. I am attaching raw log file as well as csv xls files. Thanks  

[Perf 2017-09-17 21-10-51.xlsx](https://github.com/amund7/ScanMyTesla/files/1309795/Perf.2017-09-17.21-10-51.xlsx)
[RawLog 2017-09-17 21-10-56.txt](https://github.com/amund7/ScanMyTesla/files/1309796/RawLog.2017-09-17.21-10-56.txt)

amund7, 2017-09-18:
Front RPM has a placeholder by default in the Perf tab, and should show up if the car sends it, that far I got in my investigations. Will test run the app with your logs and see if it's in there, or what goes wrong.

kdjcollins, 2017-09-18:
Thanks Amund

Ken Collins

amund7, 2017-09-18:
Fixed. The aggressive spam filter was a bit too aggressive, now has a rule to pass this packet, which has a different lenght than usual.

closed 2017-09-18 by amund7

kdjcollins, 2017-09-18:
Great. So this will be in the next release/update?

amund7, 2017-09-18:
Yep!