amund7/ScanMyTesla #17: peak min/max hp and torque values displayed

kdjcollins, 2017-09-25:
Latest release peak min/max torque values display for front and rear motor - both front & rear read the actual peak torque of the REAR motor  (in agreement with the csv log), rather than each displaying their own peak max torque values. Also, front hp and rear hp both read combined peak hp (all 3, front, rear, and combined read the same max value 635 hp, in agreement with the csv log. I attached excel file since can't attach csv and I didn't log raw. Also attached is screen shot of display at time of peak hp. It can be seen that peak front = peak rear = peak combined hp. I realize that you have a rear drive Tesla so can't see this issue on yours. Thanks!

[Perf 2017-09-25 05-32-15.xlsx](https://github.com/amund7/ScanMyTesla/files/1330131/Perf.2017-09-25.05-32-15.xlsx)
[image]

amund7, 2017-09-25:
Absolutely right, I made a mistake when converting the max/min values. Fixed, will release an update now.

closed 2017-09-25 by amund7

kdjcollins, 2017-09-26:
Works great!

Ken Collins