amund7/ScanMyTesla #22: Auto reconnect to bluetooth adapter

amund7, 2017-10-18:


closed 2017-11-03 by amund7

kdjcollins, 2017-11-19:
I seem to be getting much more frequent app freezing now, with nw “timeout” error shown on app display. On average, it now occurs every 25-30 minutes of data logging.

amund7, 2017-11-19:
But will it reconnect automatically?

kdjcollins, 2017-11-20:
No, the app is still running and trying to reconnect to no avail - it keeps trying and again reports timeout error. 

Ken Collins

kdjcollins, 2017-11-20:
Troubleshooting... I reduced the number of signals being logged to CSV file from 20 to 10. We will see if that eliminates the timeout error. 

I also connected the OBDLINK MX device to one of my ice cars (as you suggested several months ago) and rechecked for any firmware updates - it was still current. I then changed several parameters using the OBDLINK app, one of which was max timeout, which was set to 2000 msec. I also turned off fast polling. I don’t know if that only changes in that app or if it writes to the MX device. 

I ran two data logging sessions with the car on but stationary in my garage, logging the 10 signals. No timeout errors. The first session was 90 minutes, the second session 60 minutes. 

Tomorrow I will log again driving to & from work and I will see if it works. 

Ken Collins