amund7/ScanMyTesla #35: BMS/Battery readings from 75kW pack

splee777, 2018-01-14:
Hi, I just started playing with this program on a 2017 Model S 75 and noticed an anomaly on the battery readings. I'm trying to figure out if it's my pack or just a misread. The issue is that block 84 (last block) of the battery is continuously reading 3.997v, out of step with the other 83 blocks, which are within 5mV of each other. Is this an issue with the app misreading something for the 75kW pack, or do I really have an issue with block 84 of my battery? Thanks!

amund7, 2018-01-14:
I've heard of this on another occasion, and I believe that was also a 75. Send me a raw log from that tab, or if you're interested I can really reccommend using my other software called Canbus Analyzer, also here in github, to analyze your raw log files.

I guess the problem will be how can the app know that the last part of that last packet should be ignored, but if you send me a log I will have a look into it.

splee777, 2018-01-14:
Let me know if this log files contains what you need.... It was from the Cells tab

[RawLog 2018-01-14 15-05-38.txt](https://github.com/amund7/ScanMyTesla/files/1629791/RawLog.2018-01-14.15-05-38.txt)

amund7, 2018-01-14:
Excellent, I will look into it.

amund7, 2018-02-28:
Playing back your log, I can not make sense of this.

It says cell 84 is 5 volts (that part of the packet gives all bits as 1, which is clearly an illegal value)

The 85 (and 90, 100 afaik) are built up of 16 modules, each module 16 series, each series is 3.7 volts, total 96 modules, total 355V nominal, 403,2V full charged. We then have 32 temperature readings, T1 and T2 per each module. Makes perfect sense, and adds up.

So if I were to ignore the 84th voltage reading in your pack, that would give 83 series. Still gives 32 temperature readings. Or could this make sense if you have a software limited battery?

Do you have any more info on this, or could you find some more info on how this should be read & presented?

[image]

[image]

adamurban, 2018-04-02:
Here's a shot from the diag screen on my 75D, maybe this helps?
[image]

amund7, 2018-04-02:
Yeah, that sort of confirms my concerns. Have you tried the latest version of Scan My Tesla? It now hides the cell that used to show 4,999V, because that is clearly an illegal value.

It won't however line up with the diagnostics screen, the last cell of block 14 will not show.

Let me know if you find any more information on this issue.

appleguru, 2018-04-02:
Here's a shot from the diag screen on my 75D, maybe this helps? (reposting from the correct account...)

[image]