amund7/ScanMyTesla #37: Timeout. Killing thread.

kdjcollins, 2018-02-05:
Am I still the only one with this issue? Occurs on average every 30 minutes when logging (csv) only 7 to 11 signals, even just sitting in garage acquiring data.  Screen says “Timeout. Killing thread. Creating Bluetooth socket. Connecting Bluetooth socket. Read failed. Socket might be closed or timeout. Read ret:-1” and keeps scrolling said message.  Note that I have the recommended scan tool OBDLINK MX. I bought a new 2nd one to troubleshoot and it does same as the 1 year old one. Recovery requires power off/on the OBDLINK MX and close the Scan My Tesla app, restart both. Phone (Samsung Galaxy Note 3)  has latest up to date software and CPU load and RAM doesn’t seem maxed out. Question: on the OBDLINK MX, are there some non-standard settings (only visible using OBDLINK on an ice car) that I should set, such as fast data etc?  I think long ago on the first MX I tried this and it didn’t help. The same OBDLINK devices and phone work fine using Torque app on my Panamera Turbo & my Golf-R with many signals. Thanks a lot.

amund7, 2018-02-05:
Sorry to hear that you had to buy a 2nd one and it didn't help. I haven't had any other reports on this, but I do see it myself sometimes. Sometimes it gets so bad the MX seems to lose the pairing (or the phone loses it, not sure which), and I have to pair them again.

I had a theory that you have the dual motor and auto pilot car, correct? While I have an older non-AP RWD car. My theory was that your CANBUS may have a lot more traffic than mine, and that kills/overwhelms the OBDLINK MX somehow. We are using it outside it's intended purpose, so there is a possibility that some traffic and/or the app's commands are doing something the OBDLINK engineers never thought of.

There is not much I can do about this, but if you have the chance of getting a Raw log up until it stops working I could take a look and see if the MX is giving some error messages or other hints as to what is going on.

I will keep looking for a better adapter if there is one, I am particularily excited about the Comma.AI Panda. A user has sent one to me, and I'll see if it can work and be supported by the app, and if it could work more reliably than the OBDLINK.

kdjcollins, 2018-02-05:
Yes, Amund, I have an AP2 car, late 2016 P100D. I will record a RAW log on the commute home tonight (or if that doesn’t freeze, will log it in the garage) to try and catch the freeze. Question - 1. when using the included OBDLINK app software on a ice car to see firmware is up to date, etc, should I be setting any non-default parameters? Or does it NOT make any difference once out of that app and running your Scan My Tesla app? 2. Is there a diagnostic I should run on phone to ensure that it is not the root cause . ie too slow? As far as I can tell the CPU’s 4 cores are not too heavily loaded, and there is available RAM and flash storage memory. I have a couple of slightly older Android phones that I attempted to pair (Samsung S4) but they won’t connect to the OBD, maybe don’t have low energy Bluetooth interface. Thanks

amund7, 2018-02-05:
1. I recently did a firmware update on my OBDLink MX via that app, had never done it and my MX is many years old, not sure how old but maybe as much as 6-7 years? So I was hoping the firmware update would fix my occational hangups, but I can't say I notice any improvements. I have never seen the parameters you mention, post a screenshot and I can probably give an educated guess if they will change anything.

2. Your phone seems high end, and I don't think CPU has anything to do with this. I suspect it could have to do with Bluetooth chip manufacturer and/or Android version having something to do with it, it seems to me that the Bluetooth is crashing or hanging somehow. S4 should pair as well? But if you have the chance, try a completely different brand of phone, just to see if the issue is still present. MX is not Bluetooth Low Energy as far as I know.

kdjcollins, 2018-02-05:
[
[image]
](url)

kdjcollins, 2018-02-05:
Here’s a screenshot after timeout issue. Thanks for the help.

kdjcollins, 2018-02-05:
By the way, the timeout messages started several months ago after an app update. Prior to that update, when OBD froze the app would freeze too.

amund7, 2018-02-05:
No I meant a screenshot of the MX app settings :)

Yeah the timeout and reconnect were introduced a while back, before that the whole app would just freeze if it lost connection with the adapter. I realize there is still polishing to do, the app should have a notification icon that tells the user it is still trying to connect etc.

kdjcollins, 2018-02-05:
OK my mistake. I will get the screenshots of the MX on one of the ice cars using the OBDLINK app. Thanks.

kdjcollins, 2018-02-06:
I tried but didn’t get adapter to freeze and app to timeout last night when normal driving & logging 11 signals to both raw and CSV files. (Wonder does logging both files simultaneously slow data acquisition down enough that timeout doesn’t occur, or if i was just “lucky”. ) Then afterwards with car in garage, I logged the same 11 signals again to both raw and CSV files, periodically checking on the car to make sure displays were still on and obd logging - then got a freeze up / timeout at something less than 85 minutes. It may not be a “normal” failure since screens were dark - and weirdly - required separately rebooting both Tesla screens to recover. Something strange happened here logging and car itself powering off screens could have caused the obd to freeze (it still was powered up). This probably isn’t worth investigating due to powering off screens. Then this morning I logged 11 signals to both CSV and raw files in normal driving without freeze / timeout. In the morning commute freeze up is very rare since commute time (logging time)  is short (<30 min) vs evening wherein commute and logging time is long (>60 min).

kdjcollins, 2018-02-06:
Just a troubleshooting Thought: when I get obd freeze up and app timeout while driving and logging, among the 7 to 11 signals I log are Combined HP, front HP, and rear HP. From the logs I know the front and rear hp readings are normally from different times and rarely, if ever, are taken together. The don’t appear to have nearly as high sampling rate as other signals like battery power, current, voltage. Question: are these actual CAN bus signals which you add closest nearby samples to get combined hp, or is combined hp a CAN bus signal, or neither of the above and you actually use CAN bus torque and motor rpm signals for each motor to calculate hp for each? I am curious if the timeout occurs due to non-sufficiently close front and rear motor data to calculate combined hp.

amund7, 2018-02-06:
If the car goes to deep sleep, it seems to turn off everything, including the canbus communication. Maybe even the power to the adapter? Someone said that once, but I never checked it. In any case, if the bus is quiet, the app will timeout after 30 secs without any canbus traffic, and try restart bluetooth and communications. So that would be intended behaviour of the app.

kdjcollins, 2018-02-06:
Thanks for the explanation. As a different troubleshooting avenue, tonight I will receive another new OBD  adapter, a higher end but non-preferred ELM327 Kiwi3. I originally tested the app with an old Kiwi and it worked ok, but testing was brief and certainly wasn’t statistically valid, and I didn’t compare signal timing w MX. If I could get one of my other older Androids to pair w the MX I would certainly test to see if it’s the phone issue.

kdjcollins, 2018-02-06:
If I want only “Mech Power combined (kW)” signal, do I need to ALSO select front mech hp and rear mech hp so they can be added, or is that unnecessary since as you said “Mech Power combined” in kW is read directly from the CAN bus. Thanks.

kdjcollins, 2018-02-07:
I determined that If I want only “Mech Power combined (kW)” signal, I still have to also have front mech hp and rear mech hp, else "Mech Power combined (kW)" = 0 on my P100D.  Today more significantly I logged 19 signals, including said hp signals, and OBDLINK MX froze, causing app to Timeout, after <10 minutes logging while normal driving. I was logging to both RAW and CSV files so I will attempt to attach here. Timeout seems to occurs more frequently when logging more simultaneous signals. I will log tomorrow with the hp signals.
[RawLog 2018-02-06 17-36-55.txt](https://github.com/amund7/ScanMyTesla/files/1701809/RawLog.2018-02-06.17-36-55.txt)
OK RAW file is attached (csv not supported)

kdjcollins, 2018-02-07:
[image]
[image]

amund7, 2018-02-07:
What I meant to say is that Front HP and Rear HP are canbus signals, coming in at the same time. I add them together as HP Combined, whenever Front HP comes in. It should work if you log only HP Combined, but if it doesn't that's a bug that I will have to look into.

Could you take note of what happens with the LEDs of the adapter when this happens? I think in mine, the bluetooth lamp goes out, or starts blinking, like it does when it's NOT connected to anything. If we get some consistent reports, maybe we could forward this to the ODBLINK people.

kdjcollins, 2018-02-07:
In normal operation the left LED (green “power”) and the right LED (blue “Bluetooth”) stay lit continuously and the inner 2 LED’s “OBD” and “Host” blink rapidly when logging. When the timeout / freeze occurs, the outer 2 green and blue LED’s stay lit continuously, but the inner 2 LED’s (OBD and Host) are OFF. Recovery requires the following exact sequence: shut down app AND power off the OBDLINK MX device. Restart OBDLINK and wait several seconds until both outer LED’s are lit continuously, then restart the app. [
[image]
[image]

](url)

kdjcollins, 2018-02-07:
Here’s a screenshot of the OBDLINK app setup screen using MX hooked up to an ICE car. I took this back around September 22 last year per your suggestion at the time to check and update the firmware on the MX (because of the freeze up issues that I was having when logging on Tesla) which I did (updated firmware) at [the](url
[image]
) time. See the two checked parameters - fast logging and stay alive thread. At this point I can’t remember if these were defaults or not. The brand new (2nd) MX device that I’ve been running for the last few days hasn’t been updated or had any settings modified, so I’ll have to check it on ice car w obdlink app to see if these boxes are checked or not.

kdjcollins, 2018-02-07:
See the other screenshot from when I updated the first OBDLINK MX firmware on the ice car last Sept. (see description of “fixes” of previous version 3.0, regarding CAN bus communication, fatal errors, and keep alive, etc)[
[image]
[image]
[image]

](url)

kdjcollins, 2018-02-11:
Troubleshooting OBDLINK MX settings. 
Selecting under PID Polling “Fast Polling” and “Keep Alive Enabled” makes absolutely no difference in frequency of freeze up. The default is “off”. I’ve been testing for several months both “on” [and](url) recently switched to a second OBDLINK with defaults (“off”).

kdjcollins, 2018-02-11:
More troubleshooting OBDLINK MX settings to find fix to “timeout” (OBDLINK [freeze).](url
[image]
[image]
[image]

) Under “Advanced” tab, found that “Read Timeout” default was 2000 ms (2 sec).  Tried changing to 20000 ms (20 sec). Does not help. When communication loss occurs, it just increases the delay (to 20 sec) before the Scan My Tesla app recognizes the issue and begins reporting ““timeout”.

kdjcollins, 2018-02-11:
Troubleshooting Obdlink MX. Currently trying “Disable Mode 09” (reading VIN periodically) and “Do not clear input buffer”. 
[image]
[image]

amund7, 2018-02-11:
Thanks for the effort!

I think all these settings are just for that app.

Scan My Tesla has a timer of 30 seconds of not receiving any data, before it does the 'Timeout' sequence, and tries to restart the bluetooth communications.

It would be cool if the OBDLink people could have a look at this, and see if it could be a bug in their firmware, or possibly a bug in my app, but I am not able to find I am doing anything wrong. Could you try contacting their customer service?

kdjcollins, 2018-02-11:
What’s interesting is the timer change was the first thing that actually toggled anything.

kdjcollins, 2018-02-15:
I finally got an older Samsung Tablet paired with the OBDLINK MX and running with Scan My Tesla. This passed the test of running without “Timeout” freeze for several hours of logging (15 signals) operation. So given that positive sign, I wiped (reset to factory default) my dedicated phone that I normally run Scan My Tesla on, a Galaxy Note 3, and set it up in a minimalist way, latest operating system software but with no extra apps & disabled most of the bloatware included. This ran longer, maybe 3 hours data logging 15 signals (RAW & CSV at same time) before the dreaded “Timeout” error occurred. Back to “square one”.

amund7, 2018-02-16:
I have a test build for you, will email it. I found some old code that I have a hunch might cause this, that I removed in the test build I am sending. Please test it in the same way and see if it helps.

kdjcollins, 2018-02-16:
Thanks. I installed it and will report back.

kdjcollins, 2018-02-17:
Passed 1st test (50 minutes, logging 20 signals to both CSV & raw files). Good so far, need more data. More next week.

kdjcollins, 2018-02-20:
Logged 20 signals (csv & raw) for ~40 min while driving with no issues. However after parking, I closed the app, reopened it, and was monitoring the "ALL" tab without logging, and after 2 or 3 minutes got the dreaded "Timeout" fault and OBDLINK freeze. Screenshot attached. Same message. Also, if try to close app and reopen without power cycling the OBDLINK MX, the get the "broken pipe" messages. 
[image]
[image]

[image]

amund7, 2018-02-20:
Interesting. Or, I suppose this tells us that the old code I found is NOT causing the issue. 

Another user, David Hoffman, sees the same symptoms if he leaves the phone in the car for a while. See if you can try to provoke this error in that way. If we can find a way to trigger this error, then I can finally debug it.

kdjcollins, 2018-02-20:
I do sometimes test logging while the car is sitting in the garage to try to get the error. I leave the phone and key in the car with door open, radio & HVAC on and have to keep going back say every 10-15 minutes and keep car “on” to avoid car shutting down OBD which also causes timeout every time. Does David see the timeout without the car shutting down the obd (can bus)?

amund7, 2018-02-20:
The example he gave was parking the car & going to a shop, coming back some minutes later.

Will it crash if the car goes to sleep? Or will it recover the way it's supposed to, meaning it will reconnect by itself without you having to do anything?

kdjcollins, 2018-02-20:
Mine will “Timeout” whenever the car goes to sleep (screens go black). Once I came back to it while logging in the garage and it had gone to sleep, causing the timeout fault in the app and both Tesla screens had to be separately rebooted to recover. I think earlier in this thread I reported that incident. Very strange.

kdjcollins, 2018-02-21:
Got the "Timeout" fault while logging 22 signals to csv & raw files today while driving. It occurred after about 15 minutes.  I am attaching log file and screenshot .  
[image]
[RawLog 2018-02-21 05-36-16.txt](https://github.com/amund7/ScanMyTesla/files/1744796/RawLog.2018-02-21.05-36-16.txt)

amund7, 2018-02-21:
I checked the logs, it looks completely normal, until it stops.

I also left my phone in the car today, when I came back the car had fallen asleep, phone screen said timeout, and as soon as I woke the car up, the app would reconnect and continue as if there was never any problems.

Do you always log when this problem occurs? Raw log, CSV, or both? Keep looking for a common denominator, any way to trigger this problem, I would really like to get to the bottom of it, but it is impossible until I can trigger it myself.

kdjcollins, 2018-02-21:
With my car, phone and OBDLINK MX, once I get the timeout, it cannot recover by itself, also cannot recover by just terminating the app then restarting app. Recovery requires first power cycling OBDLINK MX, then starting app. As far as a common denominator goes, only thing I have observed is that the greater the number of signals selected, the lower the mean (logging) time between timeout faults. As I found yesterday, I don’t need to be logging to file at all, just acquiring & displaying ALL signals triggered the timeout. I can say that I have NOT ever seen the timeout fault when acquiring (and logging) less than about 10 signals (other than car falling asleep parked when logging). I normally log combined mechanical power and the individual front & rear mech power. The few times that I have NOT logged mech power, the timeout never occurred. Last week I tried only logging the individual front & rear mech power (without the addition you do in the app) and still got the timeout. Since the individual mech powers appear to be output at much larger intervals compared to other signals like battery power, voltage, current, torque & RPM, I can’t help but wonder if there’s a correlation. What happens if the mech power signals aren’t available on can bus when requested?

David990, 2018-02-21:
Hey it's David (sharkcookie.ev)
I read through most of the discussion here. I'm having the same issues. 
Some things I can confirm that have been mentioned here before. It definitely started with an update of the app. 
Unlike others, most of the times I can just restart the app and get it working again. One very few times do I actually have to reset the MX adapter. 
I also noticed that when I start logging (CSV files) the app is less likely to time out and shows the error. If I leave the app just running (without logging anything) it will error out almost all the time. I have to say though that I usually only log something when the car is still doing something. Either preheating or when I'm at a Supercharger. I guess in these situations the car is actually not going to sleep. 
I will try other devices and see if it helps.

amund7, 2018-02-21:
Thanks for chiming in David. Interesting you said after an update, could you possibly go back and find the approx date when you saw this problem the first time? If I have introduced the bug at a known point in time, I can look through my code history and see if I can find some hints.

kdjcollins, 2018-02-21:
I looked back at my earlier posts here on the issue of freeze up (before you added the timeout message) from 9-22-2017 when I first reported it. Not sure which app version that was.

kdjcollins, 2018-02-21:
I should add (after seeing David’s comment about logging CSV files being less likely for his to timeout) that I normally log CSV files and normally see the timeout fault. I have added logging raw files to the routine. I haven’t tried only logging raw files without CSV.

kdjcollins, 2018-02-25:
I updated to 1.4.2 and started to test in garage with 22 signals (displaying only, no raw or CSV logging yet) but I was recording the screen for less than 1 minute. The app froze and then got the “timeout” fault about 30 seconds later. Note the the phone itself was not frozen and responded to start/stop the video recording. The OBDLINK MX was frozen in the usual manner. Also as usual, resetting just the app doesn’t recover. Recovery required 1st power cycling the OBDLINK, then reloading the app. It raises the question in this specific test if video recording (cpu load) contributed to the freeze. (?). So I retested with same 22 signals but this time logging CSV & raw in garage for 30 minutes with no freeze up / timeout (I did record some screen video several times during the test without fail.)

kdjcollins, 2018-02-26:
Using new version 1.4.2  I logged 22 signals to raw & csv files while driving today and got  OBDLINK MX & app freeze then timeout in first few minutes. I was NOT recording screen when the app stopped updating, then I started recording in time to catch the timeout messaging. I will attach screenshot and raw file. ( following this I started logging 13 signals, which doesn't include any of the mechanical Power signals, to both CSV and raw files, and had no further freezing or timeout incident over the remaining 30 minutes of the drive. This freeze / timeout has NOT occurred when NOT logging mechanical power signals. 
[RawLog 2018-02-26 05-41-01.txt](https://github.com/amund7/ScanMyTesla/files/1759886/RawLog.2018-02-26.05-41-01.txt)
[image]

amund7, 2018-02-26:
What happens after this? Would it go to timeout after 30 sec, then loop over and over, never managing to reconnect?

If so, could you screenshot the error messages?

kdjcollins, 2018-02-26:
Attached GIF file screen recording of frozen app screen then timeout from above incident 
[image]

kdjcollins, 2018-02-26:
It never recovers. Had to kill app, power cycle OBDLINK MX, restart app.

kdjcollins, 2018-02-26:
[image]
I took a screenshot at the end of the original (not gif ) screen recording - probably is more clear. Error is "read failed, socket might be closed or timeout, read ret. -1".

kdjcollins, 2018-03-03:
Since last Monday, I have logged (raw & CSV simultaneously) 10 consecutive driving sessions totaling over 500 minutes with NO TIMEOUT fault (& associated OBDLINK MZ freeze) - so I think this has some statistical validity - by eliminating several signals. Notably, mechanical power (front, rear, and combined), torque & rpm (front & rear)- 7 signals. I have been alluding to some issue with mechanical power, at least with my P100D,  for some time now. Recall that with my car I get the timeout fault & OBD freeze on the average every 30 minutes or less, when displaying, with or without logging signals (number of signals anywhere from 10 to 22)  that include mechanical power. I had previously tested just eliminating "combined mechanical power" but still got the timeout. Also in previous tests, I had had included mechanical power (front, rear, and combined), but NOT included torque & rpm (front & rear), and still got the timeout fault every 30 minutes on average. Now, besides this timeout fault there are others with obvious root cause - (1) car going to sleep while logging when parked, (2) OBDLINK MX power loss (I have a switch), (3) loss of bluetooth communication. 
[image]

kdjcollins, 2018-04-06:
[image]
After a month of no timeout faults (>40 hours driving & logging!!) on ver 1.4.2 I started getting the fault again. I then updated to 1.4.3 and same behavior... I am getting the timeout about once a day (<2 hours MTBF). When I updated app, I also cleaned out cache, deleted the months data from memory, and rebooted phone. Still, faults about once a day.

amund7, 2018-04-06:
So 1.4.2 worked perfectly, and the error came back as you updated to 1.4.3? Or did 1.4.2 stop working?

In the previous post, you said something interesting, power loss/switch off of the ObdLink causes the error?

kdjcollins, 2018-04-06:
1.4.2 worked perfectly for a month, even with the suspected power signals, then started faulting every day after that good month. After several days of faults on 1.4.3 I updated app to version 1.4.3 and the faulting every day continued. 

Ken Collins

kdjcollins, 2018-04-06:
Regarding the power loss, yes, if the OBD looses power, then it induces the timeout. (I have a switch to allow one-handed reset of OBDLINK MX in the event of timeout. I accidentally bumped the switch off and induced the timeout). 
Ken Collins

amund7, 2018-04-06:
Hi Ken,
When you turn off the adapter, or unplug it, or walk out of bluetooth range, the timeout is intended behaviour, and the purpose is to try to reconnect to the adapter, if the connection is lost for any reason. The app will do this every 30 seconds until it succeeds. When you turn off your adapter, and turn it on again, the app reconnects, right?

It is very interesting that  you say 1.4.2 works perfectly. I will send you that version again by e-mail, and you can try running it for a while more to see if it stops your trouble. If we are 100% sure the problem is gone in that version, I will revert the bluetooth code to match 1.4.2.

kdjcollins, 2018-04-06:
Amund,
1.4.2 worked well for a month+, until it started faulting. Then it repeatedly faulted. So at that point I updated to 1.4.3.  I thought when 1.4.2 started faulting frequently again that perhaps my phone was getting low on memory, so I cleaned cache and deleted all the old files at the same time that I updated to 1.4.3. And then 1.4.3 exhibited (and continues to exhibit) then Dane behavior. (All operating system software was already up to date). 

Ken Collins

kdjcollins, 2018-04-06:
Sorry for typo, meant 1.4.3 exhibited “same behavior”.

amund7, 2018-04-06:
Haha, funny Autocorrect :)

Which phone are you using? I noticed that both you and David use Samsung equipment. I have a (very vague) hunch that maybe Samsung's bluetooth somehow crashes after a while. I will try to get a Samsung phone to test with myself. In the meantime, if you have the chance, see if you can test with another brand and see if there's any difference.

kdjcollins, 2018-04-06:
Amund,
Mines a 4-5 year old Samsung Galaxy Note 3, powerful in its heyday, but I keep around running  just for your app! Unfortunately my other 2 primary phones are iPhones. 😞 I’ve tried a Samsung tablet with same result back a few months ago. 

Ken Collins

kdjcollins, 2018-04-29:
Amund, I have been running the special version 1.4.2 for a week. Other than timeout faults occurring from switching tabs to quickly, today I got the first timeout fault while logging a drive. As screenshot shows, I was logging 20 signals, including mechanical power, front, rear, combined (kW units). All other units selected as English, not MKS (metric). Got the timeout after a little over an hour driving session. Same as all previous timeout faults on earlier 1.4.2 and 1.4.3.  Tried recovery by just closing app and restarting, app without power cycling OBDLINK MX. Did not work. Then power cycled MX without restarting app (while it's still looking for Bluetooth signal). After a short delay, this recovered communication and data logging. I tried to upload RAW file (was logging RAW and CSV, as I always do) but file size too big (49MB) - limit on the GitHub appears to be 10MB. Let be know if you want me to e-mail you link to file (dropbox).

[image]

amund7, 2018-04-30:
Interesting. Please e-mail me the RAW log. I saw something interesting in one log the other day - adapter said 'CAN ERROR'.

But before you power cycled - did the app still work, just repeatedly posting 'timeout'? Or was it locked up with a gray screen?

I have an idea that I could try changing the adapter reset command to a 'harder' reset, perhaps that could revoke it if it's somehow fallen into a 'can error' hangup, but is still responding to app commands.

kdjcollins, 2018-04-30:
Hi Amund,
Since the file is too big for me to send by gmail (48MB) I have sent you a link to the file in my Dropbox. Please let me know if that works. Otherwise I may be able to send via work email. 

The Scan My Tesla app did not go to gray screen and continued to scroll timeout I think - so your approach to do a reset of MX device may be good (especially if it would also restart logging function). 

Please see the link sent to your gmail address. 

Thanks Ken

Ken Collins

kdjcollins, 2018-05-03:
Amund, I sent you a dropbox link to RAW file from above referenced timeout freeze.

[8 more replies]