amund7/ScanMyTesla #38: Start automatically, log automatically, log always

strowger, 2018-02-10:
Hi,

I'm used to using Leafspy, which could be started automatically with something like Tasker, and if configured to log would then log continuously, uploading the logs to Dropbox or a server every minute.

I'd like ScanMyTesla to log continuously in the same way. At the moment, it I start it, I can select a tab, hit down-arrow and start logging, but when I leave the app and return logging has been disabled again.

I can deal with starting ScanMyTesla automatically but it would be really good if it could be configured to log everything automatically.

Is there any way to do this please?

Cheers

amund7, 2018-02-10:
If I understand you correcly, an 'auto log' checkbox would do what you want? So when you check it, it would always log a CSV of the tab you used last? That can be done!

strowger, 2018-02-11:
Hi,

Something like that yes! Ideally it would log EVERYTHING, or be configurable to log everything - wouldn't want looking at different tabs in the UK to affect what gets logged.

Cheers

amund7, 2018-02-11:
In the UK? This app should work in any country. :)

Jokes aside, I was contemplating if it is possible for Android to start an app when it finds a specific bluetooth device, then the app could auto-start every time you get to the car. Or is that something that Tasker can do?

strowger, 2018-02-11:
Hi,

Ugh I meant UI, sorry. Though it would obviously be good if you could make the app continue to work after Brexit ;-)

I'm sure you could accomplish what you suggest with Tasker.

My ask is a bit different - I dedicate a device to just running ScanMyTesla - I'd like to be able to install it "permanently" in the car somewhere and have it just start logging either when it boots, or when the car boots. I'd like to be able to look through the UI of the app, examining the different pages, and have logging continue in the background all the time.

Hopefully this makes some kind of sense :-) It's how I used to use LeafSpy.

Cheers

amund7, 2018-02-11:
Yes it does make sense. I can not help in starting the app, another app or Android needs to do that. But I think I can make 2 options:

- Auto-start logging (when the app starts)
- Log all signals (regardless of which tab you're on, and then you should be allowed to switch tabs as well)

The first one is easy. The second is a bit more involved, I need to change the filter behaviour of the bluetooth handler, and it would need a lot of testing to make sure it works in all situations.

There are also a few issues with logging all signals, depending on which adapter you have:
- OBDLink MX hangs after a while, more common on later D/AP cars (perhaps more traffic on the bus?) One user complains that his MX will crash pretty consistently after 30 minutes. To get it working again, you need to unplug the adapter, and my experience I sometimes need to reboot the phone as well.

- ELM327 devices will miss a lot of packets, and it seems to hit the hardest on the rarest packets. This includes at least the Odometer packet, which breaks all trip meters, and why the trips tab have their own special logic to cherry pick the packets it needs. 

I suggest you do some trials, try logging All for a while and see how it works out for you.
I think probably a better solution is that you create your own custom tabs that contain exactly what you want to log.

strowger, 2018-02-15:
Hi,

Thank you very much for the reply, it makes good sense. Sorry for taking so long to reply.

I have the ELM327 and it's OK so far (on a pre-facelift MS). Losing the trip meters doesn't worry me too much. I think I do want "all". I'm doing some logging and writing a script to parse the logs.

Is there a way to get absolute timestamps from the log lines? It looks like the log lines start with a "time" field but it appears to start from zero at the beginning of the file - ideally we'd have something like a Unix timestamp/epoch there - can I safely calculate that any way, for example from the ctime/mtime on a completed logfile?

Cheers

amund7, 2018-02-15:
File name has the starting timestamp of the file / 0 timestamp, you should be able to add those together I suppose.

I can almost guarantee you that with ELM and logging ALL you will never get odometer, probably not HVAC packets, probably rarely horsepower/du amps etc, but feel free to do some tests, and let me know what you find.

strowger, 2018-02-21:
Hi,

Thanks again for this.

With ELM and ALL I get the battery odometer every minute or two, but the "trip distance" only 2 or 3 times in an hour.

The temperatures outside/inside and floor vents usually every few seconds, but the values of which vents are enabled (floor/mid/window) every second.

I'm getting "torque", "input power", "stator amps", "hp" for both motors every few seconds at worst when the vehicle is being driven.

Cheers

amund7, 2018-02-21:
That's a lot better than my tests, maybe your ELM is a lot faster than the 2 I have. In my tests Battery Odometer would NEVER be recieved. That's the reason for all the stupid logic in the Trips tabs, that has to cherry-pick each packet to initialize the tab.

You should get the trip distance and battery odometer always at the same line in the CSV, because they both come from the same CAN packet.

strowger, 2018-02-21:
I don't know about my ELM, it is really old, and I did read that the Chinese makers of them keep making them cheaper and nastier.

I am not sure if I will continue to use the "all" tab or a custom tab. Either way, the feature of having logging enabled on startup, and perhaps of being able to select a tab (for example my custom one) to be automatically opened on startup, would be useful.

amund7, 2018-02-21:
I will add a 'auto log' option for starters, should be relatively easy.

amund7, 2018-04-06:
It's on my list.

Please tell us more about influxdb and grafana?

strowger, 2018-04-07:
Your app writes CSV format logs.

Dropsync uploads them to Dropbox.

I wrote some awful perl (but it's a matter of perspective, I share a desk at work with a man who regards me as a hopeless deviant because I don't use FORTRAN) to parse the CSV logs from your app and feed them to influxdb.

Influxdb works with Grafana.

And that gives stuff like this:

[image]

strowger, 2018-07-24:
"Start automatically" is easily handled with Tasker.

It looks like you've added "log automatically", "log always" in 1.5.0.

I think you've completed this so I'm closing it. Thank you.

closed 2018-07-24 by strowger