amund7/ScanMyTesla #39: Must close then restart app after copy, paste and move parameter up or down, else signals dont report

kdjcollins, 2018-02-20:
After  copying a parameter from ALL tab and pasting in another tab, then long-pressing said parameter to select, then moving parameter up or down, all signals stop reporting. There is no error indicated. Closing app then reopening app fixes reporting. It is not necessary to restart OBDLINK MX, all led's are normal. Maybe this is just a procedural requirement rather than a bug (? ). This isn't a new issue but I noticed it occurs every time.

amund7, 2018-02-20:
This one I can not reproduce. Does it only happen with a specific parameter? Can you reproduce it over and over, or was it a one-time occurance?

kdjcollins, 2018-02-20:
I’ve seen it several times and I was copying different parameters each time. Today I was copying rear stator current then moving it. Could be just the selecting then moving operation. Not really sure it it required the additional signal. I will check and just move some things around without copying and pasting.

amund7, 2018-02-20:
If you could do a screen recording, that would be helpful.

kdjcollins, 2018-02-20:
Sure, I will try that tonight.

kdjcollins, 2018-02-21:
I shot a video tonight of a copy & paste operation, then rearrange order of parameters and couldn't get the issue to recur. Will try again later.

kdjcollins, 2018-02-21:
[image] I was able to recreate the issue which may just be my operational issue not an issue with the app. And here I have attached a recorded video which had to be converted to GIF from MP4 since MP4 apparently isn't supported, so my audio narration is gone. After moving a parameter, I click the next tab (trip) to deselect  (clear) the parameter, then I click back on the new tab before trip has completed what it is loading. Following that action the new tab won't report any signals, as I show by driving reverse then forward then rotate steering wheel both directions without response. The app has to be closed and reopened to recover, but the OBDLINK MX does NOT need any action. I duplicated this sequence twice with same result. What isn't clear in retrospect is whether moving a parameter up or down has anything to do with it. Maybe just the moving too quickly between tabs triggers the issue.

amund7, 2018-02-21:
It looks like you briefly clicked the Trip tab, then back to the new tab. This seems to be an issue with the app trying to load the trip parameters, which it continues to do after you swap tabs while that part is working. What is actually happening is the app is still stuck in the trip tap because of the lengthy 'getting trip parameters' sequence. When done, it sets the filters of the trip tap, thinking it is still there, meaning the adapter will only recieve the trip related items. Had you had any trip related items in this tab they would update. I will file this as a separate bug. Thanks for reporting, and recording!

closed 2018-02-21 by amund7