amund7/ScanMyTesla #40: user manual: CSV log time and how it works..

AndKe, 2018-02-21:
What is the time field unit in CSV log ?
When I enable CSV logging.. does it run from that point on, or store data the app collected before logging as well ?
When does the log end/resume ? Indefinitely as long as the app is running ?

amund7, 2018-02-21:
Time unit is milliseconds since log start (filename will give the starting time and date)
Logg is written directly as packets come in, meaning it will only write the data recieved after you start the log.
Log runs until you stop it, or close the app. New log = new file, with new time/date in the filename, time column starts over from 0 ms.

AndKe, 2018-02-21:
Thanks, so if I drive, then stop and get out of BT range, then resume the drive, the timestamps will show that lost time ?

amund7, 2018-02-21:
Yes, if the app keeps running (and keeps trying to connect), you would - at least in UDPLogger, just see straight lines thru the 'gap' in data.

Does that behaviour seem OK to you, or would you rather have a new file when the app reconnects?

AndKe, 2018-02-21:
It's not of high importance, as it's easy to merge CSV files, however, it would be preferable to just create a new file to prevent monster sized files if the feature is left on, such a "monster file" (containing many days) would require some math just to find one particular time-span , if desired..

Also; is the "view" menuitem merely a file open dialog, or is it supposed to somehow present the data ? -Pressing it just opens some folders, but does not offer to open location of the logs, or the root of internal storage.

amund7, 2018-02-21:
Do you mean you want a maximum file size, then start a new file?

I am aware the 'view' option is quite stupid. I found a black hole in the Android API, where it seems to be impossible (in the later Android versions) to actually open this dialog with the correct folder. It's the best I can do currently, I've tried several times do fix it, but it seems the whole internet does not know how to do it. Not sure if it would be better to remove it altogether?
The good thing at least is that once you find your folder, it will remember it, and from there you can share and send any log file.

AndKe, 2018-02-21:
No, I mean start a new file on reconnect - as you suggested.

If you see any use for "View" , you could rename it to "open" (which does not imply that the user gets to see anything)  - but there is no real reason to keep it.

amund7, 2018-02-21:
I plan on making a feature to play logs back - at least the raw logs - then I would build a custom list of the log files. I think the old one will stay like it is until then.

closed 2019-02-09 by amund7