amund7/ScanMyTesla #41: Feature request: please allow loading cached data and work on gauges/tabs offline.

AndKe, 2018-02-21:
It would be very nice to have a "offline mode" , where one could work on tabs/vars in peace, not only when connected.

you could provide a default dataset with the app, but once all values are fetched, let the last state on disconnect become the dataset.

Then add an "offline" or "editing" mode.

amund7, 2018-02-21:
Yes! It is planned, but not sure how to do it yet.

1. Play back logs (yours, or a default log), like Scan My Tesla Demo does. Upside, easy, downside, can easily confuse users with believing the data they see is from their car, when it's not.

2. The app actually has placeholders. These are invisible, but gets populated when packets come in. I have been contemplating visualizing these, so that you could actually customize tabs when offline. This is somewhat important as we get more & more different cars, for instance in my car without 4x4, the app still looks for the front DU packets, and wastes some bandwidth doing so. If we could see it, we could a. delete anything not relevant for our cars, but also b. learn about what's available for other cars.

3. If you actually want to see your data, I've thought of a sort of a 'save snapshot' function. For instance, at the end of a long trip, you could 'screenshot' all your data, with the possibiltiy to read back that static data, and browse through the tabs. Just a thought, not sure how complicated it would be.

I suppose in the end I'd like to see all of these implemented, or a better idea that keeps the pro's of all. I'd like to hear your (and other user's) opinions on these. These are all a bit clunky, so my dream come true would be a better idea that solves all these in 1 go.

amund7, 2018-02-21:
Also, 
4. Save a snapshot of current data when exiting the app. When the app is restarted, all these are shown, even if there is no connection to the car.

While this one solves a few things, I really don't like the fact that it would be confusing for users to know if the data is fresh, or from last week so to speak. Maybe color it all grey?

AndKe, 2018-02-21:
yes, it should be clearly marked as "demo" data, identified by color, or an background that says "demo/offline"  

The best option - for offline editing, would be to use actual data from user's vehicle, like I mentioned, maybe store value as you see them 
init like this;
voltagedemo=360  //default demo voltage, stored in config
voltage=voltagedemo  //(init gauge if offline mode is enabled)
then on app exit, store voltage into "voltagedemo" variable.

amund7, 2018-02-21:
The problem with that approach is, I don't know if your car has a front drive unit, for instance, so with that approach a RWD car would always show static random/demo values for front HP, Nm, RPMs etc.

AndKe, 2018-02-21:
what about showing all on first start, then , on exit having a bool for "seen front_drive_unit" set to false if not seen, storing a "snapshot" , omitting everything about front drive unit ?

amund7, 2018-02-21:
That would mean that every user getting the app would see lots of false data. I think that might clutter up my inbox a bit :)

AndKe, 2018-02-21:
hence the "offline/demo" mode. 
a  background with well visible "demo mode"  - or "offline data" would prevent confision

AndKe, 2018-02-21:
another option is to skip all demo data, just let user (in offline mode) see values populated during operation..

AndKe, 2018-02-21:
[image]

amund7, 2018-02-21:
I am thinking even more obvious, when in offline mode there are no numbers, just the description.

amund7, 2018-02-21:
Oh and why don't we ALWAYS have offline mode whenever the app is not actively receiving data from the car? Meaning right on startup before anything is recieved, and whenever you lose bluetooth connection/timeout? Or would that be too much, should you need to select it manually, what do you think?

AndKe, 2018-02-21:
yes, that was kind of what suggested here https://github.com/amund7/ScanMyTesla/issues/41#issuecomment-367438625
..by suggesting to display "demo" data, and store them for next application start upon disconnect.

This way one would have a demo/offline mode , presenting last data, whenever the app started, with no data connection.  
two changes since then:
-maybe disregard injecting default data, just build it up on first run.
-offer to clear all data, (in case when a user connects to a different VIN, like a 4WD owner connects to a RWD tesla, - then he would need to clear some data...)