amund7/ScanMyTesla #43: BMS .cvs column order

AndKe, 2018-02-23:
The column order is rather strange.. 

Time | Cell temp min | Cell temp avg | Cell temp max | Cell temp diff | Cell min | Cell avg | Cell max | Cell diff | Cell  1 voltage | Cell  2 voltage | Cell  3 voltage | Cell  1 temp | Cell  4 voltage | Cell  5 voltage | Cell  6 voltage | Cell  2 temp | Cell  7 voltage | Cell  8 voltage | Cell  9 voltage | Cell  3 temp | Cell 10 voltage | Cell 11 voltage | Cell 12 voltage | Cell  4 temp | Cell 13 voltage | Cell 14 voltage | Cell 15 voltage | Cell  5 temp | Cell 16 voltage | Cell 17 voltage | Cell 18 voltage | Cell  6 temp | Cell 19 voltage | Cell 20 voltage | Cell 21 voltage | Cell  7 temp | Cell 22 voltage | Cell 23 voltage | Cell 24 voltage | Cell  8 temp | Cell 25 voltage | Cell 26 voltage | Cell 27 voltage | Cell  9 temp | Cell 28 voltage | Cell 29 voltage | Cell 30 voltage | Cell 10 temp | Cell 31 voltage | Cell 32 voltage | Cell 33 voltage | Cell 11 temp | Cell 34 voltage | Cell 35 voltage | Cell 36 voltage | Cell 12 temp | Cell 37 voltage | Cell 38 voltage | Cell 39 voltage | Cell 13 temp | Cell 40 voltage | Cell 41 voltage | Cell 42 voltage | Cell 14 temp | Cell 43 voltage | Cell 44 voltage | Cell 45 voltage | Cell 15 temp | Cell 46 voltage | Cell 47 voltage | Cell 48 voltage | Cell 16 temp | Cell 49 voltage | Cell 50 voltage | Cell 51 voltage | Cell 17 temp | Cell 52 voltage | Cell 53 voltage | Cell 54 voltage | Cell 18 temp | Cell 55 voltage | Cell 56 voltage | Cell 57 voltage | Cell 19 temp | Cell 58 voltage | Cell 59 voltage | Cell 60 voltage | Cell 20 temp | Cell 61 voltage | Cell 62 voltage | Cell 63 voltage | Cell 21 temp | Cell 64 voltage | Cell 65 voltage | Cell 66 voltage | Cell 22 temp | Cell 67 voltage | Cell 68 voltage | Cell 69 voltage | Cell 23 temp | Cell 70 voltage | Cell 71 voltage | Cell 72 voltage | Cell 24 temp | Cell 73 voltage | Cell 74 voltage | Cell 75 voltage | Cell 25 temp | Cell 76 voltage | Cell 77 voltage | Cell 78 voltage | Cell 26 temp | Cell 79 voltage | Cell 80 voltage | Cell 81 voltage | Cell 27 temp | Cell 82 voltage | Cell 83 voltage | Cell 84 voltage | Cell 28 temp | Cell 85 voltage | Cell 86 voltage | Cell 87 voltage | Cell 29 temp | Cell 88 voltage | Cell 89 voltage | Cell 90 voltage | Cell 30 temp | Cell 91 voltage | Cell 92 voltage | Cell 93 voltage | Cell 31 temp | Cell 94 voltage | Cell 95 voltage | Cell 96 voltage | Cell 32 temp

amund7, 2018-02-23:
They are arranged this way to match the physical modules, you can see it in the BMS tab.
The BMS tab was recently introduced to mimic the BMS diagnostic screen in the car, the numbering and naming of these are one of the earliest public domain/open source decodings of the Tesla can-bus.

How would you change this for the better? Keeping in mind that the community has been referring to this naming convention for about 4 or 5 years.

AndKe, 2018-02-23:
I would *not* change it :)
I was afraid this was some kind of randomness based on last clicked/moved item in GUI or something like that..

Are you familiar with http://www.nongnu.org/dataexplorer/  ?
DataExplorer would be a great way to graph/display the collected data in a meaningful way. 
it can even do, and display, calculations if needed (like "result = dataa*datab")

amund7, 2018-02-23:
I agree with you that it looks messy, it was better before the BMS tab, then they would show voltage 1-96, then temperature 1-32. But you see the dilemma of changing or renaming it now :)

DataExplorer looks very interesting! Do post any interesting graphs you make with that, maybe do a post about it in the facebook page ?

AndKe, 2018-02-24:
-I have not yet made a "device" for this kind of .cvs file.
-nor do I use facebook :)  (worked with nework security for years)
-I will publish, and let you know should I make a "device" for DataExplorer

AndKe, 2018-02-25:
> And then I realized it would always be a losing battle to try to build a compatible 'device'
Yes, you would need something like this:
[CSV2Serial2.zip](https://github.com/amund7/ScanMyTesla/files/1756386/CSV2Serial2.zip)

I generated most of the XML fields from "Battery" cvs using python  - but did not get it to work - at least yet.

> Better to spend time improving https://github.com/amund7/UDPLogger 
well, thats a windows thing.  If we agree we are not in the 90's anymore, a webservice would make much more sense.  The least would be to have the tool be multiplatform capable, like python or Qt. :)

closed 2018-04-23 by amund7