amund7/ScanMyTesla #44: .cvs format error

AndKe, 2018-03-03:
the application produces cvs logs like: 

```
Time,Battery voltage,Battery current,Battery power,DC-DC current,DC-DC voltage,DC-DC input power,DC-DC output power,Fr drive power max,Rr regen power max,Rr drive power max,Nominal full pack,Nominal remaining,Energy buffer,SOC,Usable full pack,Usable remaining,DC Charge total,AC Charge total,Charge total,Discharge total,Regen total,Discharge cycles,Charge cycles,Battery odometer,Max discharge power,Max regen power,Rated range,Typical range,Full rated range,Full typical range,Cell temp min,Cell temp avg,Cell temp max,Cell temp diff,Cell min,Cell avg,Cell max,Cell diff,
123,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,0.3538,1.342,3.5868,3.233,3.84056,3.89502791666667,3.96683,0.12627,
```

Every line ends with a comma before CR/LF
the format should be 
col1,col2,col3
data,data2,data3

no trailing commas.

amund7, 2018-03-05:
I think this is a perfectly legal CSV format, same as in Excel there are lots of empty cells. Googling it, people are actually complaning that lots of parsers (including Java built-in parsers) are ignoring that empty column (when they wanted it included).

According to who's CSV specs is this an error?

AndKe, 2018-03-05:
yes, I know, pandas and another parser I tried handled it perfectly as well, it's just unusual to see every line end with a comma, and ... its wasting some bytes :)
mainly an cosmetic issue, please close if it's hard to fix.

AndKe, 2018-03-05:
BTW: it's not about empty cells(only the redundant commas on each line)

closed 2018-03-05 by amund7

AndKe, 2018-03-05:
since you asked, and I knew that your your app produces trailing commas:
The standard is RFC4180
https://www.ietf.org/rfc/rfc4180.txt

2.  Definition of the CSV Format

   While there are various specifications and implementations for the
   CSV format (for ex. [4], [5], [6] and [7]), there is no formal
   specification in existence, which allows for a wide variety of
   interpretations of CSV files.  This section documents the format that
   seems to be followed by most implementations:
```

   1.  Each record is located on a separate line, delimited by a line
       break (CRLF).  For example:

       aaa,bbb,ccc CRLF
       zzz,yyy,xxx CRLF
```

in light of that, it may be better to fix it.