amund7/ScanMyTesla #45: Heating/cooling power value includes supercharging power

strowger, 2018-04-22:
When supercharging, the power received is reported as part of the "heating/cooling" total.

Car is a prefacelift 70D.

Issue happens consistently - it's not just a single glitch value, I have a log full of it.

strowger, 2018-04-22:
[image]

amund7, 2018-04-22:
I know, we don’t have a reading for charge rate, so can’t subtract that. 

This should probably be explained somewhere… I am contemplating documenting and/or commenting every single reading.

From: strowger
Sent: søndag 22. april 2018 20.13
Subject: [amund7/ScanMyTesla] Heating/cooling power value includessupercharging power (#45)

When supercharging, the power received is reported as part of the "heating/cooling" total.
Car is a prefacelift 70D.
Issue happens consistently - it's not just a single glitch value, I have a log full of it.

amund7, 2018-04-23:
But, the real solution would be to find a 'charge rate' reading, I have one I found somewhere on the net, but haven't yet gotten around to test it. Will look into it.

strowger, 2018-04-23:
I don't know if there is a frame on CAN3 which has the data, but you might be able to infer it: 

Speed zero and/or no motor torque/rpm/power and cell voltages rising?

When supercharging is not limited (eg by adjacent stall in use, or connector overheating), the charge rate and the BMS' "max charging power" seem to be identical. 

Or, a bad way to do it, you could call the Tesla API like the official app does, and fetch charging status and voltage/current.

kdjcollins, 2018-05-12:
Heating/Cooling power appears to include additional power signals well beyond the 5-6 kW heater. It is evident while driving with battery heater and climate control off. Min/Max typically reports >50kW, and fairly high "negative" power as well from normal driving. Check out this screen shot - typical.
[image]

closed 2019-05-26 by amund7