amund7/ScanMyTesla #50: Please modify CSV format by appending new values to end, or at least documenting the changes

strowger, 2018-07-24:
Hi,

Thank you for the new release! I love that you have found more values to report/log. It is great to have more understanding of the coolant flow and temperatures.

Please would you consider adding the new values to the end of the CSV lines when you modify the CSV file format?

I have written code to parse the log lines and it is becoming increasingly ugly and complex as you modify the file format over time, and I have to deal with the different versions.

It would be nice if you would append all new values to the end of the log line.

If that does not appeal, please could you document the log format change in the changelog when you make a new release? I noticed that you swapped the floor vent & mid vent values around at some point which confused me.

Cheers

amund7, 2018-07-26:
I am trying not to change the names of items because of this. But if your code can't deal with changing orders, you need to write more robust code.

The order of things are due to the placement in the default tabs, and they may change every now and then, for layout reasons. The order in the logs could be sorted differently, but how? Alphabetically? Then the order would also change if something is inserted or removed later on, right?

Oh maybe you can fix it yourself, by reordering the items in your tab? I can't remember right now if that would also change the order of the CSV columns, or if they stay fixed, but you could try that.

strowger, 2018-07-26:
Thank you for the reply. My code is crap, no doubt about that.

I use the "All" tab, always have done.

amund7, 2018-07-26:
Please don't take it that way, but with CSV, excel, also databases, the *index* of the column is often/always subject to change.

In some of my excel parsing tools for work, I usually have a lookup-list of known column headings I am looking for, then fill the indices into a hasmap/lookup on program startup. That works pretty well.

closed 2019-02-09 by amund7