amund7/ScanMyTesla #51: SMT seems incompatible with Google Nexus 6 handset

strowger, 2018-09-17:
Hi,

I've been using SMT successfully for many months with the ZTE Blade A110, which runs Android 5.1.

I tried to use it with a Google Nexus 6 device. This is a more powerful and larger device. I successfully paired it with the ELM327 and was able to run "TM-Spy" perfectly. When I start Scan My Tesla, I am able to select the Bluetooth device, but it never connects. It sits forever at "read failed, socket might be closed or timeout, read ret: -1".

I have tried rebooting ELM327 and phone. I have tried deleting the bluetooth pairing and re-making it. I have tried the last stock OS made for Nexus 6 (Android 7.1.1 build N6F27M), and the latest nightly of Lineage OS (which is Android 8). SMT will not work but "TM-Spy" works reliably.

Any ideas how to make Scan My Tesla work, please?

amund7, 2018-09-17:
I am baffled on this. Could you try to capture a Raw log and send me? Maybe we could get some hints as to what's going on.

strowger, 2018-09-17:
Hi,

Thank you for the quick reply.

I tried again with all the logging turned on: CSV, Raw/debug, Full CAN dump, and auto rec on start.

It produces zero-length CANDump files.

It produces All*csv files which are either zero-length, or contain only the first line (with the field labels in).

I see no raw logs.

```
-rw-rw----  1 root sdcard_rw    0 2018-09-17 19:41 CANDump 2018-09-17 19-41-32.txt
-rw-rw----  1 root sdcard_rw 3072 2018-09-17 19:42 All 2018-09-17 19-41-32.csv
-rw-rw----  1 root sdcard_rw    0 2018-09-17 19:43 CANDump 2018-09-17 19-43-26.txt
-rw-rw----  1 root sdcard_rw 3072 2018-09-17 19:43 All 2018-09-17 19-43-26.csv
drwxrwx--x 17 root sdcard_rw 4096 2018-09-17 19:44 ..
-rw-rw----  1 root sdcard_rw    0 2018-09-17 19:48 CANDump 2018-09-17 19-48-28.txt
-rw-rw----  1 root sdcard_rw    0 2018-09-17 19:48 All 2018-09-17 19-48-28.csv

```
Cheers

strowger, 2018-09-20:
Hi,

I tried with only raw/debug selected and it produced a zero-length log.

I watched it for a few cycles of attempting to connect, and tried restarting the app, but the log is definitely zero bytes long.

closed 2019-02-09 by amund7