amund7/ScanMyTesla #65: Questions about 1.6.1 log format

strowger, 2019-04-23:
Hi,

Not sure what the best place for this sort of question is, pointers appreciated. Have also asked on Facebook "wall" but the post seemed to vanish.

First of all - thank you for continuing to update SMT. It's great software and I love it.

In 1.6, we now have a logfile item "battery heater temp" - is this the same as the old "coolant heater exit" item?

What are "battery heater req" and "battery heater state"?

What are the new "thermal load" items? I understand that you wrote "thermal controller load" in the changelog but I don't really understand what this means.

Cheers

amund7, 2019-04-23:
I found them here, but could never agree with myself how to name them.
https://github.com/apach3guy/CAN3/blob/master/thc.py

Right after releasing them, I realized that 'thermal load' means the opposite of what I am trying to convey... more correct would be 'thermal controller load' or 'thermal controller consumption'. Suggestions for better names are welcome.

PS. Sorry that Facebook does this, I don't get any notification if someone posts there, and it's very hard for anyone to find.

strowger, 2019-04-24:
Hi,

Great, thanks for this. Good news about the old "heating/cooling" number, which as you noted on facebook is mostly nonsense. Will keep replies to here, rather than Facebook.

Am I right about "battery heater temp" - it's the item which used to be called "coolant heater exit"?

Cheers

amund7, 2019-04-24:
Yes, that is correct.

closed 2019-04-28 by amund7